marcoskirsch/nodemcu-httpserver #2: Only works once

marcoschwartz, 2015-02-28:
Hello, I am having an issue with the web server on my ESP8266 boards. It works fine for the first request, but then the server is not accessible anymore & I need to restart the chip. Do you know if this is a known issue? Thanks

marcoskirsch, 2015-02-28:
I just uploaded a new version of the code, give that a shot. Let me know how it goes.
Questions:
1) What build of the firmware are you using?
2) Do you have floating point enabled in the build?

marcoschwartz, 2015-03-03:
Works better now, even if I still have some problems for example serving files. I am using 0.9.5, and to be honest I don't know about the status of floating point enabled.

From now how could I not serve file, but for example JSON to build something like an API running on the ESP chip? Thanks

marcoskirsch, 2015-03-03:
_to be honest I don't know about the status of floating point enabled_
Then you probably have floating point enabled.
          
        print(0.2+0.1)

...will print 0.3 if fp is enabled.

_but for example JSON to build something like an API running on the ESP chip?_
The MIME type is "application/json". You can add extension json it to the list of MIME types and I will merge it later. If, more likely, you want to return json from your server side script, then just make sure to use this MIME type when you send the HTTP header. Let me know how it goes.

closed 2015-03-03 by marcoskirsch