marcoskirsch/nodemcu-httpserver #8: Post

zhivko, 2015-03-14:
Is it possible to post like form, and fetch some text entered in textbot in lua variable?

marcoskirsch, 2015-03-14:
Right now the form must use GET, not POST. Check out args.lua example.

bhasden, 2015-03-31:
Just a heads up, but I added the support to parse headers and url encoded post requests in my fork of the project. Like Marcos mentioned, the memory is very tight, but I needed the ability to do some sort of form posting and simple url encoded post handling was easy to implement.

Marcos, I'm not sure if you're accepting pull requests or would rather manually merge in changes, but if you need anything let me know and feel free to grab whatever you'd like (if anything at all) if you feel it's useful.

Oh, and thanks for putting this project together. It's been super useful as a starting point for serving files from the ESP8266 devices.

pastukhov, 2015-03-31:
Brian, did you try to make discover access points first?
I am getting out of memory :(
https://github.com/pastukhov/nodemcu-httpserver/blob/master/http/updateaplist.lua

bhasden, 2015-03-31:
pastukhov,

I'll take a look tonight to see if I can discover access points. Just a guess, but the use of the json serializer may be causing your issue. It's possible that the list of APs that are found is causing the cjson module to use up all of the available memory. Have you tried retrieving the list of APs without serializing the result to try to determine exactly when you're running out of memory?

At this point, I've pulled down a copy of the nodemcu firmware and planning to build a version without some of the modules (i2c, pwm, etc.). I'm hoping that frees up enough memory to be able to do a couple of other things outside of running the web server.

marcoskirsch, 2015-03-31:
This is great. As soon as I can I will merge your changes. Thanks for improving the project for everyone!

On the memory issue: I wish the builds had all the extra features disabled by default rather than the other way around.  Oh well.

pastukhov, 2015-04-01:
Brian, no. Any call of wifi.sta.getap() causing reboot.
```lua
local function storelist(t)
end

local function sendHeaders(connection,args)
 connection:send("HTTP/1.0 200 OK\r\nContent-Type: application/json\r\nCache-Control: private, no-store\r\n\r\n")
 connection:send('{"error":0, "message":"OK"}')
end

return function(connection,args)
 wifi.sta.getap(storelist)
 sendHeaders(connection,args)
end
```

bhasden, 2015-04-02:
pastukhov,

I'm noticing the same thing regarding any call to wifi.sta.getap() causing a reboot. I can run your script by itself (with minor changes to run in init.lua) and it runs just fine, but when run from within the webserver page serving pipeline it's causing a crash. 

BTW, I'm now running a custom version of the nodemcu firmwire with most things disabled in order to free up memory, so it doesn't appear to be a memory related issue.

I was hoping to create a setup page that provides a drop down list of available APs for the user to select before entering a password, but that doesn't appear to be an option at this point. Looks like I'll have to figure something else out.

I would really appreciate it if you reported back if you happen to figure out how to retrieve an AP list while a server is running.

Good luck!

pastukhov, 2015-04-02:
I have an ide to run wifi.sta.getap in background via tmr and store as http/aplist.json but not sure about filesystem multiple write stability and concurent run with http server.

pastukhov, 2015-04-09:
At least this code is working for me
```lua
tmr.alarm(2, 1000, 1, function()
    conn = net.createConnection(net.UDP, 0)
    conn:connect(1234,"192.168.100.122")
    conn:send("1234")
    conn:close()
    conn = nil
    node.dsleep(5*1000000)
end)
```

marcoskirsch, 2015-04-19:
@bhasden did you ever get this working? Pull request would be nice!

bhasden, 2015-04-20:
Yes, url encoded POST requests work properly in my fork of the source code. For some reason, my line-endings aren't matching (seems to be the same issues pstukhov was having), so a pull request is going to be a bit difficult. I see you've committed some updates recently, so I'll try to get them pulled into my branch, resolve any conflicts, test the POST code with the updates, and send you a pull request sometime this week.

marcoskirsch, 2015-04-20:
Ok sounds good. Thanks!

marcoskirsch, 2015-07-19:
@bhasden would love to get your pull request that adds support for POST.

simcop2387, 2015-09-02:
Not sure what happened with @bhasden 's branch but I also needed it and I've got a PR here: https://github.com/marcoskirsch/nodemcu-httpserver/pull/34

It passes the request table to the lua files so they can decide on what to do with a given method.  It does however keep the request body and headers around in memory so it might not be perfect.  I've thought about changing it to parse one or two headers and keep only that around and then just keep the body of the request around until you ask it to parse it.

I'm also thinking of changing it to not restrict the method type at all in httpserver.lua so that you don't have to change that either but with most of the methods I didn't see them as being particularly useful anyway (CONNECT to name one at least).

Let me know what you think and I'll make changes.

closed 2016-02-16 by marcoskirsch