marcoskirsch/nodemcu-httpserver #9: http GET request from our software

tcpipchip, 2015-03-15:
After our IOT software request a HTTP GET for your HTTP SERVER, we got the following error
https://dl.dropboxusercontent.com/u/83554849/bug_http.txt
Please, see if you find out the bug.
Miguel

marcoskirsch, 2015-03-16:
From the looks of it you are just requesting the root folder. This works fine for me with the example index.html I include.
> Does it work for you with the index.html I include?
The server runs very tight, memory wise. Using a version of the firmware with no floating point helps. 
> What version of the firmware are you running?
> Do you have floating point turned off?

I just uploaded some bug fixes, please try the latest version of the server.

tcpipchip, 2015-03-16:
Hi, does not work the index.html when i can from our IOT software request. But works using local ip.
I am using now nodemcu_integer_dev_json_20150317.bin
I think that the float is turning of.
In the LAN, i use CHROME.
Can you say me wich version you use there ?

tcpipchip, 2015-03-16:
here the link
https://dl.dropboxusercontent.com/u/83554849/error.txt

marcoskirsch, 2015-03-16:
> I think that the float is turning of.
Yes, if it says integer then there's no floats.

I last tested with nodemcu_integer_dev20150311.bin

tcpipchip, 2015-03-16:
can you give me the link for download ? I will test and report you! Thanks

marcoskirsch, 2015-03-16:
> PANIC: unprotected error in call to Lua API (?:0: attempt to concatenate field 'uri' (a nil value))

There's only a few places in the code in which a variable uri is concatenated. You should be able to figure out which one is breaking and put the appropriate protection around it. It's probably easiest for you to try it since you can reproduce the crash. Then send me a pull request.

Thanks!

marcoskirsch, 2015-03-16:
>can you give me the link for download ? I will test and report you! Thanks

https://github.com/nodemcu/nodemcu-firmware/releases
But I don't think it's the firmware's problem, more likely a bug in the code.

tcpipchip, 2015-03-17:
Hola Marco
Well, i have updated the firmware but i still have the problem. Look the image with the new index.html
https://dl.dropboxusercontent.com/u/83554849/marco.txt
Strange that error. There is a URI :(
Btw, can you put your esp8266 as a server in your side (public ip e port) ? Then i can do a proxy and access your ESP8266 for you see our GET request from our system in your side. I have the skype for we syncronize some day.
Gracias
Miguel

tcpipchip, 2015-03-19:
Marco, can you explain what  this function do ?

local function parseRequest(request)
   local e = request:find("\r\n", 1, true)
   if not e then return nil end
   local line = request:sub(1, e - 1)
   local r = {}
   _, i, r.method, r.uri = line:find("^([A-Z]+) (.-) HTTP/[1-9]+.[1-9]+$")
   return r
end

marcoskirsch, 2015-03-19:
Sorry I haven't had a chance to sit down and debug this.

That function takes the request that arrives from the browser, stores the first line in variable e, then it breaks it up so that r.method contains the method (i.e. GET) and r.uri contains the URI (i.e. "/"). I think this is what's failing for you.

Good thing is this function can be copied and run locally on your computer using a Lua interpreter. Much faster than debugging in the chip itself.

tcpipchip, 2015-03-19:
I find out the problem.  I am using the HTTP 1.0 :)
Now is working! I can access your WEB SEVER from anyware from the WORLD :).
Btw, i want that you page be official for webserver when we have the product in the market.
Btw, have you fully tested your web sever for bugs ?

marcoskirsch, 2015-03-19:
Yeah makes sense.
This should fix it:
   _, i, r.method, r.uri = line:find("^([A-Z]+) (.-) HTTP/[1-9]+.[0-9]+$")

Try it out and send the pull request.

> Btw, i want that you page be official for webserver when we have the product in the market.
Awesome!

> Btw, have you fully tested your web sever for bugs ?
It's still a work in progress. There are still many improvements that can be made. I want to clean up the code, reduce memory usage, etc. There are also features I'd like to add. Of course any outside help is appreciated, that's why the code is open source and published in GitHub.

I am interested in what you're working on, send me info if you want to.

tcpipchip, 2015-03-20:
Yes, i had already changed and fix it.
It´s working now. Yes, the memory is critical. Strange that sometimes give fail memory (i am not more changing the code).
Would be cool if you try to reduce the code.
Well, the idea is put 20 esp8266 in my house and share them to the internet cloud and make access to their webpage using internet cloud. The esp8266 connect to AP with special software that make the 20 esp8266 visibile on the WAN.

tcpipchip, 2015-03-20:
[image]

marcoskirsch, 2015-03-22:
Fixed. Please try it, but I remind you that you can make changes yourself to the codebase and then submit a pull request so that the fix makes it to this repository and everyone can benefit.

https://github.com/marcoskirsch/nodemcu-httpserver/commit/84f7a06aa104d402c81e2873116ed886b609b380#diff-4b40021cc36f6d8b6622fed5208d5769

closed 2015-03-22 by marcoskirsch

tcpipchip, 2015-03-22:
Thanks! Why in your example in the file list you find files that begins with /HTTP ?

tcpipchip, 2015-03-22:
By the way, did you try to make many RELOADS in a few times you will got a Memory panic error !

marcoskirsch, 2015-03-22:
Leading slash (/) means that files are in the root of the file system. 

What do you mean by "RELOADS"?

tcpipchip, 2015-03-22:
Reloads of webpage in the browser many times in a second

marcoskirsch, 2015-03-22:
Ok please open a separate issue for that.

tcpipchip, 2015-04-01:
http://esp8266anywhereweaved.blogspot.com.br/2015/03/weaved-fabric-for-internet-of-things.html

Reference to your WEB SERVER

marcoskirsch, 2015-04-01:
Cool! thanks for sharing.

tcpipchip, 2015-04-01:
your web server is the better one!

tcpipchip, 2015-06-29:
Marcos
Did you see that they did the HEAP memory 2 times bigger ?

marcoskirsch, 2015-06-29:
@tcpipchip I did see! I was hoping to test it this weekend, but nfortunately I had an Internet outage AND I have a bricked NodeMCU I need to bring back from the dead. If you try it out please share the results.

tcpipchip, 2015-06-30:
I will test tomorrow! Ok ?

tcpipchip, 2015-06-30:
wIll you do upgrades ?

tcpipchip, 2015-07-01:
Hello Marco
I tested your web server in the LUA version that have more HEAP.
nodemcu_integer_0.9.6-dev_20150627

Node info

majorVer: 0
devVer: 6
chipid: 10016474
flashid: 1261768
flashsize: 512
flashmode: 0
flashspeed: 40000000
node.heap(): 17960
Memory in use (KB): 13
IP address: 192.168.8.108
MAC address: 18:fe:34:98:d6:da

It´s working. I think that now you have more space for good upgrades.

tcpipchip, 2015-07-01:
I forgot to tell that i had to reduce the PNG images. Looks that this LUA version is reaching near 512Kb :(