marcoskirsch/nodemcu-httpserver #10: not enough memory

zhivko, 2015-03-21:
I am getting:
NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
set mode=STATION (mode=1)
MAC: 	18-FE-34-9F-E0-A5
chip: 	10477733
heap: 	20032
lua: init.lua:15: not enough memory

What is the minimum heap for server to run?

marcoskirsch, 2015-03-21:
I have not run it on builds with floating point enabled due to memory constraints. Try an integer-only build. This is in the ReadMe. 

I think it's possible to do a little bit of refactoring in the server code to improve memory usage. Just haven't gktten to it.

zhivko, 2015-03-21:
LUA_NUMBER_INTEGRAL is defined in
line 80 in my
user_config.h

Will try that thanks.

marcoskirsch, 2015-03-23:
I'm about to upload a slightly refactored server. This now works with floating point enabled but I must warn you that memory is pretty tight, can't serve very large files :(

marcoskirsch, 2015-03-23:
Fixed, please try this again.

https://github.com/marcoskirsch/nodemcu-httpserver/commit/75ed944678404535f1685629e57f5a6c73c8b58a#diff-19e2537a7e22b7b1909faea16eccb764

closed 2015-03-23 by marcoskirsch