marcoskirsch/nodemcu-httpserver #12: Add ability to serve gzip compressed content

pastukhov, 2015-03-26:
Many js and css libraries are greater then available space but fit in case of  gzipped.

marcoskirsch, 2015-03-26:
This sounds like a good feature to add. I don't know much about how this works but I imagine you want to upload somelib.js.gz and server needs to send that with the MIME for .js and some extra info to let client know contents are compressed?

pastukhov, 2015-03-27:
Yes. Looks like server should add proper header for .gz
Look at http://www.cyberciti.biz/faq/linux-unix-curl-gzip-compression-test/ and http://en.wikipedia.org/wiki/HTTP_compression

pastukhov, 2015-04-01:
I am working on it. Almost done, but i'm in stuk with yor regexp that getting extension from file name.
In my case extension of index.html.gz should be html.gz, but with your regexp i'm getting gz. Can you advice?

marcoskirsch, 2015-04-01:
I would try and open the file with .gz and if that fails then without. Use that information to know if the file you need to serve is compressed.

pastukhov, 2015-04-02:
I need to rewrite 
```lua
   _, r.ext = r.file:match("(.+)%.(.+)")
```
to something, that return all from the first point to end of file name. At this moment it return all from last.
I wrote some changes to httpserver-static.lua that detect if file extension ends with .gz, set compressed flag and cut .gz

pastukhov, 2015-04-08:
Implemented

closed 2015-04-08 by pastukhov

marcoskirsch, 2015-04-19:
@pastukhov When I try to upload everything using

     make upload_all

the file system fills up and I enter a reboot loop (cannot open init.lua for read or something along those lines). If I get rid of the sample jquery.min.js.gz you added the problem no longer occurs. It's a fairly large file for ESP8266 standards at ~35 KB. I have an ESP-01 and to tell the truth don't even know how much flash space I have on it.

Are you able to upload the server plus all the examples under http/ fine? Thanks!

pastukhov, 2015-04-19:
Yes, same problem. Http folder does not fit to 512k version.

pastukhov, 2015-04-19:
I propose to remove jquery stuff.

marcoskirsch, 2015-04-19:
I was thinking the same but wanted to confirm. I still want a .gz example so maybe I'll add a simple compressed html or something.

marcoskirsch, 2015-04-19:
Hey @pastukhov I have another question. As it stands, if you want to request a compressed file the URL must include the .gz compression. Are you sure that's how it's supposed to work? I was under the impression that you request the file without the .gz extension and the server would know to send you the compressed file (most servers compress server-side and send which obviously nodemcu-httpserver cannot do).

pastukhov, 2015-04-20:
Great idea!
So, our server should serve  http/file.gz  if it exist. If not, it should serve http/file.

marcoskirsch, 2015-04-20:
Exactly.