marcoskirsch/nodemcu-httpserver #15: http requests HEAD and POST causing reboot

pastukhov, 2015-04-03:
Requested URI: /
Error: 400: Bad Request
PANIC: unprotected error in call to Lua API (?:0: attempt to index a nil value)
.
.
.
PANIC: unprotected error in call to Lua API (attempt to index�?)��19@H
)��4!����

pastukhov, 2015-04-03:
Marcos, are you sure that
```lua
            if req.methodIsValid then
               if req.method == "GET" then onGet(connection, req.uri)
               else dofile("httpserver-static.lc")(conection, {code=501}) end
            else
               dofile("httpserver-static.lc")(conection, {code=400})
            end
```
from httpserver.lua is right code?

marcoskirsch, 2015-04-03:
I'm sure it used to work originally, but the server code has been modified since. Now that I think of it, look like we need a req.methodIsSupported entry on the req table so we can differentiate errors "bad request" from "not supported"

CosmicMac, 2015-04-14:
It looks like a typo:
conection vs connection

raz123, 2015-04-17:
@CosmicMac, good catch.

marcoskirsch, 2015-04-17:
I'll try to get to it this weekend unless someone beats me to it (pull requests?)

marcoskirsch, 2015-04-19:
I fixed the typo plus a few other things, but that didn't fix the bug. I'm a little puzzled:

In httpserver.lua when the server gets a POST request, function onReceive is called, as expected. Eventually we figure out that POST isn't supported and call the function defined in "httpserver-error.lc". I have verified that this happens correctly and the function is indeed called with the right parameters. But the chip restart afterwards anyway!

I'd appreciate if someone can try this out using the latest code (I just posted some changes). I also added a post.html that makes it easy to reproduce the issue. It may be something really silly that I'm missing.

Thanks!

marcoskirsch, 2015-07-18:
Re-tested with a nodemcu build that has more free RAM and this works now. Closing.

closed 2015-07-18 by marcoskirsch