marcoskirsch/nodemcu-httpserver #30: WiFi configuration is not ready to operate as AP only

pospanet, 2015-08-09:
init.lua is not ready to set wifi mode to AP only. IP setting is missing at all and init is forced to get IP as client, otherwise http server will not start

marcoskirsch, 2015-08-09:
What you say is correct. init.lua bundled with the server is meant to be edited to suit the client's need and it just happens to be written so by default it joins an existing network. I don't think it's an issue. Am I missing something?

pospanet, 2015-08-09:
Hi,
if it is like "to be modified", then yes, you are right.I made some
modification whe roy need to amend "just" WiFi mode and the rest Will be
handled with code. Modification is ready to be merged. I am going to send
pull request and it is up to you to decide if it is useful or not. Thanks.

Pospa

S pozdravem
Jan Pospíšil

-<{Pospa}>-

closed 2016-02-17 by marcoskirsch