marcoskirsch/nodemcu-httpserver #32: Cancelled requests fills up memory and causing reset

hazarkarabay, 2015-08-14:
Output from serial console attached below. Heap size printed every 100ms using:

    tmr.alarm(2, 100, 1, function()
    	print(node.heap())
    end)

I'm refresing file_list.lua page in my browser every 400-500ms. Important thing is page loading time slower than my refreshing speed - which effectively means cancelling previous request and making new request every time. These cancelled requests do not free its resources, will add up and causing resetting.

If catching these half-completed and cancelled requests possible, this issue can be fixed.

	26768
	Requested URI: /file_list.lua
	18280
	20040
	20040
	18216
	Requested URI: /file_list.lua
	14400
	15624
	15624
	14696
	Requested URI: /file_list.lua
	10840
	12072
	12048
	11112
	10800
	Requested URI: /file_list.lua
	7216
	8512
	7640
	7800
	Requested URI: /file_list.lua
	3616
	4888
	4888
	4000
	4176
	PANIC: unprotected error in call to Lua API (not enough memory)
    ~~ reboot and bootloader gibberish ~~ 

Sorry for my poor English and I hope the issue above understandable.

marcoskirsch, 2015-08-14:
Off the top of my head, I'm not sure how this can be solved. Let me think about it or maybe someone has an idea.

IntelliDust, 2015-12-08:
Slow response is caused by many dofile() during one query.

try to use
````lua
tmr.alarm(2, 100, 1, function()
    collectgarbage()
    print(node.heap())
end)

PR #50 (2016-02-15): Fix a memory leak when the connection is dropped by the client side
Fixes #32
closed 2016-02-15 by marcoskirsch