marcoskirsch/nodemcu-httpserver #36: Crash after POST request from iOS & MacOs Safari

NicolasFOVET, 2015-09-09:
Using the post.lua example from Safari web browser (on both iOS and MacOS) leads to a crash of the server. No problem using other web browser or cURL tool with POST method.

Here's the log : 

Requested URI: /post.lua
Method: GET
Requested URI: /post.lua
Method: POST
Getting Request Data
Parsing Form Data
PANIC: unprotected error in call to Lua API (?:0: attempt to index a nil value)
PANIC: unprotected error in call to Lua API (attempt to index a nil value)
PANIC: unprotected error in call to Lua API (attempt to index a nil value)
PANIC: unprotected error in call to Lua API (attempt to index a nil value)
PANIC: unprotected error in call to Lua API (attempt to index a nil value)
...
...
...
PANIC: unprotected error in call to Lua API (attempt to index a nil value)
PANIC: unprotected error in call to Lua API (attempt to index a nil value)
PANIC: unprotected error in call to Lua API (attempt to index a nil value)
PANIC: unprotected error in call to Lua API (attempt to index a nil value)
PANIC: unprotected error in call to Lua API (loop in gettable)
PANIC: unprotected error in call to Lua API (attempt to concatenate a nil value)
PANIC: unprotected error in call to Lua API (attempt to call a string value)

PS: anyway, let me thank you for your great work, this server is just awesome ! :)

marcoskirsch, 2015-09-10:
POST support was added recently. My guess is that it's a simple parsing bug. I am out of town right now so I can't look into it soon. Hopefully someone gets to it sooner.

Thanks for the nice words!

anyn99, 2015-12-25:
Hi there, 
had the same problem here. But on an android device. The problem is, that the mobile browser apparently split up the POST request into two tcp packets: a header without a body, and the body is sent immediatly afterwards.
here my solution, tell my if it also fixes your problems:

in httpserver.lua add the following after "local function onReceive(connection, payload)":
----------------
            -- if header says there is content, but this content is missing inside body of packet
            if payload:find("Content%-Length:") then
                if (tonumber(string.match(payload, "%d+", payload:find("Content%-Length:")+16)) > #payload:sub(payload:find("\r\n\r\n", 1, true)+4, #payload)) then
                    tmp_payload = payload
                    return
                end
            end
            
            if tmp_payload then --add payload to former payload if body was missing
                payload = tmp_payload .. payload
                tmp_payload = nil
            end
-------------

leave the rest of the function as is.

best wishes and nice holidays,
Daniel

anyn99, 2015-12-27:
Hi again,

the proposed code in my last post is pretty rough and only collects data from one more incoming data packet, if it is missing.
The following code, inserted at the same position, is prettier and collects packets until their size matches the size that is stated inside the header.
the variable tmp_payload adds up data, so it could use up all heap and cause a reboot if the packet is too big. May that be a problem? I don't know a way to prevent that, maybe someone else can improve?
-----------
-- collect data packets until the size of http body meets the Content-Length stated in header
                  if payload:find("Content%-Length:") or bBodyMissing then
                    if tmp_payload then tmp_payload = tmp_payload .. payload else tmp_payload = payload end
                    if (tonumber(string.match(tmp_payload, "%d+", tmp_payload:find("Content%-Length:")+16)) > #tmp_payload:sub(tmp_payload:find("\r\n\r\n", 1, true)+4, #tmp_payload)) then
                        bBodyMissing = true
                        return
                    else
                        print("HTTP packet assembled! size: "..#tmp_payload)
                        payload = tmp_payload
                        tmp_payload, bBodyMissing = nil    
                    end
                  end
--------------------

devyte, 2016-01-25:
I ran into this same issue, and was starting to fix it. I'm glad someone already found a solution.
Maybe a pull request should be opened?

marcoskirsch, 2016-01-26:
Yes please open a pull request.

marcoskirsch, 2016-02-16:
Update:

I just did a good amount of cleanup in the server code, and things seem to be running but this issue is still there. I confirmed it works on Firefox.

I also want to change how the POST is handled so arguments are passed in to script in the args variable rather than req variable to match how things are done in GET.

marcoskirsch, 2016-02-16:
Used suggested snippet, just pushed the fix. Thanks!

closed 2016-02-16 by marcoskirsch