marcoskirsch/nodemcu-httpserver #37: Base64 decoding fails on nodemcu float builds

hazarkarabay, 2015-09-16:
Base64 decoding gives unexpected results on nodemcu builds with floating point support.

**nodemcu_integer_0.9.6-dev_20150704.bin**
```
NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
> print(math.floor(1.5))
stdin:1: malformed number near '1.5'
> print(dofile("httpserver-b64decode.lua")("dGVzdGluZw=="))
testing
```

**nodemcu_float_0.9.6-dev_20150704.bin**
```
NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
> print(math.floor(1.5))
1
> print(dofile("httpserver-b64decode.lua")("dGVzdGluZw=="))
tUstYnf
```

hazarkarabay, 2015-09-16:
```
print(string.format("%.20f", 2^4), "\n")
> 16.00000000000000355271

print(string.format("%.20f", 2*2*2*2), "\n")
> 16.00000000000000000000	
```
which means
```
print(2^4 == 16)
> false
```

This floating point rounding weirdness causing this issue. Wrapping ```math.round()``` around every exponential calculation in b64decode.lua fixes it, but I don't think that is a "solution". 

I suggest posting a note to readme about this, because currently this project does not working on float builds. (Basic HTTP authentication enabled in default config and logging in is impossible with correct credentials thanks to this issue)

Closing now and will try to ask nodemcu guys about this.

marcoskirsch, 2015-09-20:
This is unfortunate. 
Another option is to use the base64 module if available (C implementation) and fallback to the Lua one if not available. I'm not a fan of requiring the module and I am also not a fan of having the two different code paths. 

One thing we could do is error out/warn in the case we detect floating point bios of the firmware. This would save someone time. 

I'll leave reopened until we do any of these. I am not sure there's anything that can be done from the firmware's side.

hazarkarabay, 2015-09-20:
I actually opened an issue at nodemcu-firmware and here is the response what I got:

> This is an IoT chip, so an error of 1 in 2^51 is hardly important. If you are that time critical then either select an integer build or use the explicit form. I think that anyone involved in the firmware development would classify this as not worth fixing.

I'm now switched using an exponentiation by multiplication method that works only positive integer exponents (which base64 algorithm do use) like this:

```
function uipow(a, b)
    local ret = 1
    if b >= 0 then
        for i = 1, b do
            ret = ret * a
        end
    end    
    return ret
end
```

and changed every ```^``` operation to ```uipow``` calls. It even runs ~%30 faster on float builds based off my benchmark.

[image]

The modified version can be found at https://gist.github.com/hazarkarabay/4f641796e56a8462bc80

marcoskirsch, 2016-02-16:
FYI: nodemcu-firmware plans on adding this. See https://github.com/nodemcu/nodemcu-firmware/issues/844

I think that will be a better solution in the long run.

marcoskirsch, 2016-12-26:
I've made the changes you suggested since a C implementation in the firmware hasn't made it.
https://github.com/marcoskirsch/nodemcu-httpserver/commit/10bcd2f170714259984350ce9816a9a6760a53a5

closed 2016-12-26 by marcoskirsch