marcoskirsch/nodemcu-httpserver #40: init.lua error

claudioferrari, 2015-10-12:
Hello,

I have this problem after make upload_server:

init.lua:55: cannot open/write to file

I'm using nodemcu_float_0.9.6-dev_20150704.bin.

Any suggestions?
Thanks in advance

marcoskirsch, 2015-10-12:
Format your flash memory and try again. The file system in NodeMCU is not great about reusing space or running out of space. If that doesn't work, then remove the largest files from the examples folder, like the images, and try again.

Let us know if that helped. Good luck!

claudioferrari, 2015-10-13:
It seems it doesn't compile files:

```
...
> dofile("httpserver.lc")(80)
cannot open httpserver.lc
....
```

marcoskirsch, 2015-10-13:
Can you confirm that you formatted and started over?

claudioferrari, 2015-10-13:
Yes, I have also tried to upload all file manually but still get this error (in Explorer):
```
NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4
set (mode=3)
AP MAC: 	1a:fe:34:9d:ad:af
Client MAC: 	18:fe:34:9d:ad:af
chip: 	10333615
heap: 	24840
Compiling:	httpserver.lua
lua: init.lua:55: cannot open/write to file
```

pastukhov, 2015-10-14:
Looks like yout spiffs is corrupted. Do file.format() and reupload all the files.

claudioferrari, 2015-10-14:
Thanks guys, I'm finally able to authenticate but I get 404 - Not Found error if I try index.html (http://<my-ip>/index.html).

Suggestions?

Thanks

marcoskirsch, 2015-10-15:
Can you confirm that:

a) the URL you typed in the browser does not have the typo that the URL in the message has and
b) that index.html is uploaded correctly in your ESP
?

claudioferrari, 2015-10-16:
Hi, I finally had a working system (not yet tested POST request).
I have fixed some "bugs" (return should be the last call otherwise has to be incapsulated into do return something end).

I am trying to modify the code in order to reduce the server charge by using ajax call to functions; next step will be to flash all the stuff in ROM by using spiff (surely you are more skilled on this than me).

If you want send me a private email and I will explain you better what I am doing.

Thanks a lot again, when I will be ready I will put into a public repo.

marcoskirsch, 2016-02-16:
Closing. @claudioferrari there's much newer server and firmware code now, worth trying again. The firmware guys have improved things so much.

closed 2016-02-16 by marcoskirsch