marcoskirsch/nodemcu-httpserver #42: Problems with nodemcu dev built with 1.4.0 SDK

pjsg, 2015-11-22:
It appears that the semantics of connection:send (as used in httpserver-error) has changed in recent builds. In particular, it appears that you can only do one :send before you have to wait for the "sent" event. Subsequent :send calls appear to fail. Not clear what the right solution is.....

marcoskirsch, 2015-11-22:
I haven't had a chance to try the new versions of the firmware. Sounds like there's some work to be done :(

pjsg, 2015-11-22:
I did submit a pull request to address it. The fix works for me, but I'm 
new to Lua so it may not be stylistically correct.

Philip

marcoskirsch, 2016-02-15:
I am merging fixes for new firmware soon. The server appears to work a lot better now - thanks to you and the rest of the nodemcu-firmware people, not thanks to me :)

pastukhov, 2016-02-15:
Sounds great!

marcoskirsch, 2016-02-16:
All pushed, try again!
Thanks.

closed 2016-02-16 by marcoskirsch