marcoskirsch/nodemcu-httpserver #47: Hangs after first connection

bepursuant, 2015-12-18:
I formatted my MCU, flashed all these files to it, and changed the configuration so it could connect to my WiFi. It connects successfully, but the first time I try to hit it with any type of request, the console registers the call (below), but the browser just hangs.

I disabled the basicauth and it functions exactly the same - no response of any kind (even status codes or headers) ever makes it back to the browser.

I've been digging through recent pulls a little bit to see if there was something that could cause this, but I haven't seen anything obvious yet. I also checked with a different power supply to make sure it wasn't hanging due to a power issue.

Can anyone confirm this on the latest master branch, or am I just crazy?

ilendemli, 2015-12-18:
nah, you are not crazy, same happens for me

marcoskirsch, 2015-12-18:
I suspect nodemcu-httpserver may need some modifications in order to again work out of the box due to changes in the newer nodemcu-firmware, which has had a lot of changes.

What firmware version are you using?

bepursuant, 2015-12-26:
My module is running 0.9.6... 

Could this be related to https://github.com/nodemcu/nodemcu-firmware/issues/846 ? I see some coroutining in the nodemcu-httpserver code near where we are getting stuck, but I don't fully understand the structure of that piece of the code so I'm not sure...

exilaus, 2015-12-30:
same here 0.9.6 last version of firmware tested on this software? 0.9.5 work right?

marcoskirsch, 2015-12-31:
I think we should get this working with the latest firmware (1.5 SDK based nodemcu-firmware) and not try to get it up and running on such old firmware code anymore. There's been a ton of bug fixes in the firmware.

rlust, 2016-02-05:
Any progress with getting this to work with newer firmware? I am using 1.4 and it hangs on browser access. This looks like a great platform to build on if I could get it to work. I like the garage door scenario, just what I was looking for!

marcoskirsch, 2016-02-15:
I have gotten the code all cleaned up and working on the latest firmware. I tested on 1.5 but I think 1.4 will work (@rlust maybe you can confirm?)

Will upload later.

marcoskirsch, 2016-02-16:
Uploaded, let me know if it's working for you. Thanks.

closed 2016-02-16 by marcoskirsch