marcoskirsch/nodemcu-httpserver #49: Cannot upload files using make upload_*

QuocurIQ, 2016-01-08:
Preparing esp for transfer.
Transfering init.lua as init.lua
Verifying...
Transfering httpserver.lua as httpserver.lua
Verifying...
Traceback (most recent call last):
  File "/usr/local/bin/nodemcu-uploader.py", line 509, in <module>
    uploader.write_file(f, d, args.verify)
  File "/usr/local/bin/nodemcu-uploader.py", line 233, in write_file
    data = self.download_file(destination)
  File "/usr/local/bin/nodemcu-uploader.py", line 179, in download_file
    if bytes_read > int(size):
ValueError: invalid literal for int() with base 10: 'connection:send(table.concat(self.data, ""))\r'
make: *** [upload_all] Error 1

marcoskirsch, 2016-02-15:
Can you use nodemcu-uploader.py on its own?

marcoskirsch, 2016-02-22:
No answer, works fine for me, and no one else has complained. Closing.

closed 2016-02-22 by marcoskirsch