marcoskirsch/nodemcu-httpserver #56: Cannot get web page to load

rlust, 2016-02-23:
I am using Esplorer to load NodeMCU 8266 12E. I have tried every thing I can think of to get the server to give me a response.

I get an IP address 192.168.1.140
I have tried several urls like
http://192.168.1.140/index.html)[http://192.168.1.140/index.html]
http://192.168.1.140/index.html

Have index.html and http/index.html loaded on the ESP.

Any ideas would be great, this looks just like what I need!

Thanks

pastukhov, 2016-02-23:
Esplorer do not have ability to load files with / in the name. Use esptool.py

marcoskirsch, 2016-02-23:
Do you get 404 error? Or no response?

rlust, 2016-02-23:
No response

rlust, 2016-02-23:
I used esplorer and changed the file name of index.html to http/index.html

marcoskirsch, 2016-02-23:
> No response

Did you start the server? init.lua has a line:

    dofile("httpserver.lc")(80)

that could be commented out. If the server is running, you'll see this in your serial connection (in ESPlorer):

    nodemcu-httpserver running at http://192.168.1.140:80

rlust, 2016-02-23:
OK some progress the dofile was commented out.'

nodemcu-httpserver running at http://192.168.1.140:80
> Connecting to WiFi Access Point ...
IP: 	192.168.1.140

But still not connecting to page.
Thanks for your help!!

rlust, 2016-02-23:
I started the http server after the wifi started and got the following error when using the following request:
http://192.168.1.140/index.html)[http://192.168.1.140/index.html]
> IP: 	192.168.1.140
    dofile("httpserver.lc")(80)
nodemcu-httpserver running at http://192.168.1.140:80
> GET: /index.html)[http://192.168.1.140/index.html]
Error 400: Bad Request

marcoskirsch, 2016-02-23:
400?
What browser are you using?

rlust, 2016-02-23:
On a mac have tried Safari, Chrome, Firefox... Have found Safari seems to have issues on ESP.
Here is my environment:
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-02-13 18:32
 powered by Lua 5.1.4 on SDK 1.4.0
set (mode=3)
AP MAC: 	1a:fe:34:d4:c2:b6
Client MAC: 	18:fe:34:d4:c2:b6
chip: 	13943478
heap: 	34648

marcoskirsch, 2016-02-23:
I do all my testing on Mac and mostly Safari. That's odd.

What firmware version do you have? Built with which modules?

And do you have the latest server code?

rlust, 2016-02-23:
NodeMCU custom build by frightanic.com
branch: master
commit: c8037568571edb5c568c2f8231e4f8ce0683b883
SSL: false
modules: file,gpio,net,node,tmr,uart,wifi
build built on: 2016-02-13 18:32
powered by Lua 5.1.4 on SDK 1.4.0

Not sure what server code you are asking about?

marcoskirsch, 2016-02-23:
This server.

rlust, 2016-02-24:
Yes running the current version of the server;...what should my url request look like?

Here is the 400 error I am getting:

NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: file,gpio,net,node,tmr,uart,wifi
 build 	built on: 2016-02-13 18:32
 powered by Lua 5.1.4 on SDK 1.4.0
set (mode=3)
AP MAC: 	5e:cf:7f:0c:24:05
Client MAC: 	5c:cf:7f:0c:24:05
chip: 	795653
heap: 	34544
nodemcu-httpserver running at http://192.168.1.139:80
> IP: 	192.168.1.139
GET: /index.html)[http://192.168.1.139/index.html]
Error 400: Bad Request

Again thanks for all of your help!

rlust, 2016-02-24:
Can the wifi ip and web server ip be different?
/

rlust, 2016-02-24:
Made some progress, I got nodemcu uploader to work!  

Still getting error 400 bad request.

What is a valid url to open index.html?

marcoskirsch, 2016-02-24:
Previously, you mentioned using ESPlorer for uploading files. In the instructions in the README it is recommended you use the included makefile which uses nodemcu-uploader. One of the reasons is that the makefile has a list of the files that need to be uploaded.

Did you upload all the server files to the NodeMCU? This is the list needed:

      httpserver.lua
      httpserver-b64decode.lua
      httpserver-basicauth.lua
      httpserver-conf.lua
      httpserver-connection.lua
      httpserver-error.lua
      httpserver-header.lua
      httpserver-request.lua
      httpserver-static.lua

rlust, 2016-02-24:
Yes, I was able to get nodemcu-uploader working and uploaded all of the files that way.

What are some url examples that I should be using?

rlust, 2016-02-26:
Any ideas guys, I am stuck and would really like to get past not being able to get a response from the web server?

This looks like a great application.
I used the uploader and I have the web server running, but cannot get a page response. The browser sets at connectting for a minute or more then finally times out.

I have tried every url I am aware of, maybe I am missing a key element in the url request.

Thanks for any help you can provide!

rlust, 2016-02-26:
I got it working!!!!
I started fresh again and have success, thanks for your assistance!

marcoskirsch, 2016-02-26:
Ok, sounds like it was a usage mistake on your part. Glad it's working and glad it wasn't a bug in the server.

If you think the instructions were lacking or something in the code can be improved, feel free to open a pull request.

closed 2016-02-26 by marcoskirsch