marcoskirsch/nodemcu-httpserver #58: init.lua failing to start server

wardg32, 2016-02-27:
Apologies for what I expect is a noob issue rather than a genuine issue but perhaps the answer will be educational to other people.

I'm using init.lua as supplied except for the obvious additions. For some reason it appears the server is attempting to start before even trying to join the network, therefore `wifi.sta.getip()` is NIL and `dofile("httpserver.lc")(80)` does not execute.

So adding this print statement

```
-- Uncomment to automatically start the server in port 80
print(not not wifi.sta.getip(),  not not wifi.ap.getip())
if (not not wifi.sta.getip()) or (not not wifi.ap.getip()) then
    dofile("httpserver.lc")(80)
end
```

results in the following output upon reset

```
NodeMCU custom build by frightanic.com
	branch: master
	commit: c8037568571edb5c568c2f8231e4f8ce0683b883
	SSL: false
	modules: file,gpio,net,node,ow,tmr,uart,wifi
 build 	built on: 2016-02-25 15:48
 powered by Lua 5.1.4 on SDK 1.4.0
set (mode=1)
Client MAC: 	18:fe:34:a4:b7:a0
chip: 	10794912
heap: 	37168
false false
> 
Communication with MCU...
Got answer! AutoDetect firmware...

NodeMCU firmware detected.
=node.heap()
41856
> Connecting to WiFi Access Point ...
IP: 	192.168.1.198
```

At that point I can manually execute init.lua again and it works fine. I can see various ways of rectifying this but before I do I'd like to understand why it's happening in the first place. Thanks!

milkpirate, 2016-04-29:
remove all the `not`'s (double `not`' dont even make sense). then it should work:

```
if wifi.sta.getip() or wifi.ap.getip() then
    print(wifi.sta.getip(), wifi.ap.getip())
    dofile("httpserver.lc")(80)
else
    print("cannot start server. we havent got any IP")
end
```

marcoskirsch, 2016-05-22:
Please submit a pull request with the recommended improvements.
Thanks!

closed 2016-05-22 by marcoskirsch