marcoskirsch/nodemcu-httpserver #59: Strip down server

milkpirate, 2016-04-29:
Hello Marcos,

I am totally in love with your nodemcu-server. I was trying to write you a mail but obviously you didnt want be reached like that :).

I have a project where I was trying to use it but with the other Lua code I need everything seems to heavy to run on my ESP. So I thought about stripping down the server to the bare minimum. I only want to serve a single file and parse some arguments from the browser. Extracting the parser  function worked quite well. The problem I am facing now, is to do multiple conn:send()'s.

I already looked it up and everything results in nasty cascading of function. Is soo convinient to do multiple conn:send()'s in the Lua files I can serve with your server. I really would like to have that for myself as well. I checked your code and you seem to use a coroutine but I couldnt figure out how exactly. I would really appreciate it if you would give me a minimal example for that.

What I would like to do is following:

> Page request

- parse arguments
- execute some code depending on arguments
- send file (remote.html)...
- while sending manipulate one or two lines of remote.html (preferablyon the fly). like so:

```
if file.open("remote.html", "r") then
	while true do
		local line=file.readline()
		if line == nil then break end
		line = line:gsub("STR-2-RPLC",str(depending on arguments from above))
		conn:send(line)
	end
end
file.close()
```

As you might know it is not that easy to understand a hugh bunch of code someone else wrote, plus I'm not a Lua guru. So I would be really glad if you could help me out with that.

Best regards

marcoskirsch, 2016-05-22:
Sorry for the delay replying.
Why do you want to strip down server? Are you running out of flash memory?
RAM wise I don't think you'll get many improvements, we've really pared it down. The coroutine thing is the best we've been able to do for getting around the nodemcu-firmware limitations.

I think you should delete all the examples, and try that.

milkpirate, 2016-05-23:
Hey, Yes it's about RAM. I rewrote the server to make like that:

```
conn:on("receive", function(conn, payload)
	if not file.open(fname) then
		print("file "..fname.." not found")
		return
	else file.close(fname) end

	local idx = 0
	
	function nextChunk(c)
		local str, term
		file.open(fname)
		
		if file.seek("set", idx) == nil then
			term = true
		end

		str = file.read(512)

		if idx == 0 then
			str = str:gsub("STR_TO_RPLC",replacement)
		end

		c:send(str)
		idx = idx + 512
		file.close()

		if term then
			c:close()
			return
		end
	end
	
	conn:on("sent", nextChunk)
	nextChunk(conn)
end)
```

Not that elegent as your solution but it works as I need it to. Thank you!

marcoskirsch, 2016-12-26:
Server seems to be running quite well with recent builds of the firmware. I'm going to close issue, since stripping down the server will always be an ongoing goal.

closed 2016-12-26 by marcoskirsch