marcoskirsch/nodemcu-httpserver #66: node_info in standalone mode

HEI-Luc, 2016-06-14:
Hi
Thank for all your work, it's usefull for me and my IOT project(s)

I want to use my nodeMCU in a nomad project, so it's configured in Station and Access Point, to be connected to the network of my house if available, or by the access point of the node itself.

I detect a problem with the node_info page, because when I'm without my home network, it's try to concat the nil IP in the string to send.

my quick patch is to check the wifi.sta.status in an IF statement before ask the IP

Thanks for all,
Luc

marcoskirsch, 2016-06-14:
Sounds good! please submit a pull request. Thanks!

marcoskirsch, 2017-06-01:
Fixed in https://github.com/marcoskirsch/nodemcu-httpserver/commit/54b6525805a80d8456ac95e2b307ad6b703deeeb by @HHHartmann

closed 2017-06-01 by marcoskirsch