marcoskirsch/nodemcu-httpserver #68: Lua Panic on uri.file length > 27

jrandombob, 2016-10-19:
In spite of what the nodemcu doco says, it appears that filenames > 27 characters (not 32) cause file.open to throw a panic.

Utterly knucklehead patch attached;
[httpserver.lua.patch.txt](https://github.com/marcoskirsch/nodemcu-httpserver/files/540347/httpserver.lua.patch.txt)

marcelstoer, 2016-11-11:
The docs to my knowledge don't say but it's 31 chars, not 32. See https://github.com/nodemcu/nodemcu-firmware/pull/1114 for details.

31 - ".lua".len() = 27?

jrandombob, 2016-11-12:
No, these files didn't have extensions.

The filenames were truncated MD5 hashes under the http/ pseudo directory.

e.g. http/5f4dcc3b5aa765d61d8327 (27 chars) worked but http/5f4dcc3b5aa765d61d8327d (28 chars) caused a panic.