marcoskirsch/nodemcu-httpserver #77: Simple text files are not UTF-8 encoded

marcoskirsch, 2017-01-28:
I think that in the year 2017, if we can support only one string encoding, it should be UTF-8.
This should apply to all text MIME types (text/css, text/html, etc).

Added example hello_world.txt for testing. By default it looks incorrect, see top window in screenshot. It should look like the bottom window.

[image]

CharlesScoville, 2017-03-07:
You need to use the "UTF-8 BOM" in *.txt file to get most things to render any UTF-8 in it.

https://en.wikipedia.org/wiki/Byte_order_mark

Can do this on Window$ by opening the *.txt in Notepad++, then use the "encoding" file menu to change to "encode to UTF-8-BOM." The nodemcu-httpserver served file will then get rendered properly.

Or at least it works for me on my phones Fire Fox browser.

[image]

 (^_^)

marcoskirsch, 2017-03-09:
Adding Byte Order Mark is a workaround to the issue, not a fix. BOM is not needed nor recommended for UTF-8. What is happening is that your browser doesn't receive encoding information in the HTTP header (the bug), but then infers it when it finds the BOM in the text file.

CharlesScoville, 2017-03-10:
Hello.

>Adding Byte Order Mark is a workaround to the issue, not a fix.

(1) I don't mean to make this a semantic dispute, but a workaround is a type of fix; a "temporary fix[.]"[[1]](https://en.wikipedia.org/wiki/Workaround) It certainly fixed the specific problem I was seeing. Did you perhaps mean it's _not the preferred solution?_

(2) It's already established that this project itself is "abuse," so I don't see the merits of quibbling that we are not using some standards body approved specification. Putting a BOM at the beginning works for this file, why is that not good _enough_?

>BOM is not needed nor recommended for UTF-8.

I know that's what the wiki article says, but, frankly, that's ten year old info[[2]](https://en.wikipedia.org/wiki/Byte_order_mark#cite_note-3) that's no longer perpetuated[[3]](http://unicode.org/faq/utf_bom.html#BOM), and in many practical cases is incorrect. It's actually a very divided topic where decisions ultimately have to be made situation to situation.[[4]](https://stackoverflow.com/questions/2223882/whats-different-between-utf-8-and-utf-8-without-bom) (see the comments)

The issues with including an initial BOM only arise for very specific situations, such as breaking the shebang in a script, being interpreted as control sequences in some antique protocols, or when concatenating two files. . . none of those circumstances apply to this (*.txt) situation, of course.

The situation that we _are_ having, however, is that there is nothing telling the receiver of the file that it is UTF-8 encoded. The UTF-8 BOM can be used for this, it's allowed.

>What is happening is that your browser doesn't receive encoding information in the HTTP header (the bug).

I see. It seems that you are saying that the server is not even attempting to tell to the browser that a file is UTF-8. Is that correct?

Well, according to W3C you should "[u]se character encoding declarations in HTTP headers if it makes sense, and if you are able, for any type of content, **but in conjunction with**[sic] an in-document declaration." [[5]](https://www.w3.org/International/questions/qa-html-encoding-declarations#httphead) They go on to say that there are "a number of potential disadvantages[.]"

So, referring back to "the preferred solution," it seems you shouldn't really be relying on the header, that info should be in the *.html file as `<meta charset="utf-8"/>`. 

And, from a separations of concerns perspective, this really makes the most sense to me. The http server should not concern itself with the character encoding, it's job is to grab the file and serve it. The encoding is a typography decision that is the concern of the web page developer.

>... but then infers it when it finds the BOM in the text file.

I respectfully disagree with the claim that the browser is merely inferring this information. "An initial BOM is only used as a signature [in UTF-8]."[[6]](http://unicode.org/faq/utf_bom.html#bom5) The only other times you could run into it in a human readable file are the absurdly improbable (more than 1 in 17 billion) data sequence that represents the BOM, and the ZERO WIDTH NON-BREAKING SPACE that was totally superseded by U+2060 WORD JOINER.

So it's not really a heuristic clue, the mark can really only be found if it was put there for this reason. That makes it an "explicit statement" which makes it not an inference by definition.[[7]](https://www.google.com/search?q=define+infer&ie=utf-8&oe=utf-8)

marcoskirsch, 2017-05-27:
@CharlesScoville ok I'k going to close this issue. Thanks for your clarifications.
We won't have the server decide the policy that text files are in some specific encoding.

closed 2017-05-27 by marcoskirsch

flip111, 2017-07-15:
What about just simply adding HTTP header `Content-Type: text/html; charset=utf-8` ?

i'm also not in favor of using BOM because it's more difficult to control (some bytes on the start of a file are more easy to miss than http header or another solution)

@CharlesScoville does the browser display the html file as utf-8 if it has `<meta charset="utf-8"/>`?

CharlesScoville, 2017-07-16:
> @CharlesScoville does the browser display the html file as utf-8 if it has `<meta charset="utf-8"/>`?

One sec...
. . .
Yes, it does.  Haven't tried it on my phone browser, but it's mobile Firefox, so it should be no different.

Anyone can confirm this themselves, too, by simply copy/pasting the utf-8 data from the text example into something like index.html, then opening that file in your browser straight from a local copy. The html files in this project all seem to have `<meta charset="utf-8"/>` already declared by default.

>What about just simply adding HTTP header Content-Type: text/html; charset=utf-8 ?

The problem is how to decide when to use this header or not. 

The way I see it, the user of the project will (rightfully) want to be the one to decide the encoding of their "website," and that will change user to user, and possibly file to file. It's not something that can/should be global and static. The only way this project could be able to be adaptive like that is if there is some heuristic or deliberate mark on user files that clue the server in on the encoding they're trying to do. 

...and that just puts us right back where we started.

>i'm also not in favor of using BOM because it's more difficult to control (some bytes on the start of a file are more easy to miss than http header or another solution)

That *is* the big trade off, yes.

I would think, however, that a clear notice of the issue combined with examples that do it right should be sufficient. This project isn't exactly used by normal users, so it doesn't *have* to be that user friendly.

Case in point:
The "folders" on nodeMCU are names containing slashes, something you can't do on Windows. I couldn't find any other way to work with this than to [fork ESplorer](https://github.com/CharlesScoville/ESPlorer) and recode it with an "upload folder" feature that renames the files in transit. That was just to get the http folder's contents on to my ESP8266. That is certainly not something normal mortals would typically do. The BOM issue is trivial in comparison, though this is probably also an extreme example.