marcoskirsch/nodemcu-httpserver #78: PANIC: unprotected error in call to Lua API

ramzilla-shell, 2017-02-17:
Hello @marcoskirsch,

Your code is awesome! it connected to my wifi and it shows it has access as a client, however i cant seem to load the index.html. im using the ff:
-esplorer
-d1miniPro
Can you help me with this, I was reading your read me and you said that the filename should have a prefix "http/" but how do i do this cuz i know filenames can't have slashes. hope to hear from you sir, you have a good day!

Thanks!

marcoskirsch, 2017-02-17:
Did you upload all files using:

    make upload_all

?

If so, when the chip restarts and the server starts, it will give you the URL needed in order to connect to the chip. Your client (laptop, phone, whatever) needs to be in the same WiFi network as the ESP, of course.

ramzilla-shell, 2017-02-18:
Hi @marcoskirsch,

Thanks for the prompt reply, nope i manually loaded the files. do you have http and cjson on your module?

marcoskirsch, 2017-02-18:
The http and cjson modules are not need by nodemcu-httpserver.

ramzilla-shell, 2017-02-18:
oh i see, im sorry about this im new with esplorer and i dont know where to input the "make upload_all" :(

marcoskirsch, 2017-02-18:
If you are on macOS or Linux, go to the nodemcu-httpserver directory in the terminal and type the command "make upload_all".

If you are on Windows, then I don't know how to get GNU Make up and running and it's probably not worth the hassle anyway. You then need to upload all the server files manually in ESPlorer.

I'm going to close this issue since it isn't clear to me that you've found a bug. If you need more help getting up and running, it would help if you pasted the output of ESPlorer when you reset the chip.

closed 2017-02-18 by marcoskirsch