marcoskirsch/nodemcu-httpserver #97: Support protecting only certain pages using Basic HTTP Authentication

marcoskirsch, 2017-07-02:
Right now it's all-or-nothing.
I would like to have the ability to only use HTTP Authentication on some pages but not others. I imagine this by having a regular expression that matches the pages to protect in httpserver-conf.lua.

flip111, 2017-07-15:
A router would be a nice addition. It maps incomming url's to scripts instead of executing scripts directly. This way you can also control which middleware to run (such settings some hooks voor http basic auth).

flip111, 2017-07-15:
A router would be a nice addition. It maps incomming url's to scripts instead of executing scripts directly. This way you can also control which middleware to run (such settings some hooks voor http basic auth).

that is also related to https://github.com/marcoskirsch/nodemcu-httpserver/issues/81