marcoskirsch/nodemcu-httpserver #102: Root directory is cluttered. Move files to a folder.

marcoskirsch, 2017-07-03:
When this project started, all the server code was in a single file. Now it's spread across lots of little files, all in the root directory:

httpserver-b64decode.lua
httpserver-basicauth.lua
httpserver-compile.lua
httpserver-conf.lua
httpserver-connection.lua
httpserver-error.lua
httpserver-header.lua
httpserver-init.lua
httpserver-request.lua
httpserver-static.lua
httpserver-wifi.lua
httpserver.lua

I'm toying with the idea of reorganizing and moving all the server files into a folder. I'd remove the prefix to each file. The folder may be named httpserver/ or simply server/

I'm not about to go and do this, as it can be pretty disruptive to other contributors. So I'm floating the idea here in order to get some feedback.

Thanks!

HHHartmann, 2017-07-03:
While on the one hand that seems a good Idea on the other hand it is difficult if the files are named differently on the computer and the ESP. 
It works fine with the install script but when you use a different tool to transfer files you always have to rename them by hand.
I think a naming schema as now is fine.

marcoskirsch, 2017-07-04:
How do you transfer files to the NodeMCU?

ATAMAH, 2017-07-04:
I am using this code in init.lua:

`local compileAndRemove = function()
	for name in pairs(file.list()) do
		local isHttpFile = string.match(name, "^(http/)")
		local isCompiled = string.match(name, ".+(\.lc)$")
		local isLuaScript = string.match(name, ".+(\.lua)$")
		
		if (name ~= 'init.lua') and (name ~= 'init_start.lua') and (name ~= 'LLbin.lua') then
			if (not isHttpFile) and (not isCompiled) then
				if isLuaScript then
					if file.open(name) then
						file.close(name)
						file.remove(string.sub(name, 0, -3) .. "lc")
						print('Compiling:', name)
						node.compile(name)
						file.remove(name)
					end
				end
			end
		end
	end
end

local renameAndRemove = function()
	for name in pairs(file.list()) do
		local isHttpFile = string.match(name, "^(http/)")
		local isServerFile = string.match(name, "^(httpserver)")
		
		if (name ~= 'init.lua') and (name ~= 'init_start.lua') and (name ~= 'LLbin.lua') then
			if (not isHttpFile) and (not isServerFile) then
				if file.open(name) then
					file.close(name)
					print('Moving:', name)
					file.remove("http/"..name)					
					file.rename(name,"http/"..name)
					file.remove(name)
				end
			end
		end
	end
end`

It automatically compiling all ".lua" files and move all files named not "^httpserver*" to "http/". Maybe somebody can make PR from this code.

HHHartmann, 2017-07-04:
@marcelstoer currently I use ESPlorer. I don't want to install python on my (windows) machine right now.
I would like to switch to ftp some time.

I also have a similar script to rename the http files after upload, but not as sophisticated as @ATAMAH s
It has an explicit list of files, as I am also storing other files in the root directory.

marcelstoer, 2017-07-04:
@HHHartmann did you mean to tag @marcoskirsch rather than me?

CharlesScoville, 2017-07-18:
>How do you transfer files to the NodeMCU?

I've customized my fork of ESPlorer to give a "Save Folder" button specifically so I could work with this project. It also has a dist dir with a prebuilt jar, for those who don't want to download the entire SDK and compile.

:3