marcoskirsch/nodemcu-httpserver #108: Does default mimetype should not be "text/plain"?

Godzil, 2017-09-04:
Should will it be better to use a mime type like "application/octet-stream" than "text/plain" for unknown types?

https://github.com/marcoskirsch/nodemcu-httpserver/blob/28fbb6e7a7b7efe06dac092bbf0a8f8c1ae9c799/httpserver-header.lua#L18

I know that we are not going to serve really complicated thing with this server, but if the file is not text we will probably get corrupted data as the browser will expect 7bit ASCII

marcoskirsch, 2017-09-05:
I don't have a strong opinion on this. I honestly don't know what's best. Ideally nobody relies on the default.

If you serve a H or CPP or PY file from the server, you'd want text. What would you serve where you'd want octet-stream?

marcoskirsch, 2017-09-05:
Ok. I learned to use Google.
I think your suggestion makes sense. Feel free to do a PR.