marcoskirsch/nodemcu-httpserver #111: Why not using file.exists instead of file.open()+file.close()?

Godzil, 2017-09-05:
In plenty of location there are construction like:

https://github.com/marcoskirsch/nodemcu-httpserver/blob/28fbb6e7a7b7efe06dac092bbf0a8f8c1ae9c799/httpserver.lua#L64
https://github.com/marcoskirsch/nodemcu-httpserver/blob/28fbb6e7a7b7efe06dac092bbf0a8f8c1ae9c799/httpserver.lua#L65

where a file is open; then close to check if the file exist.

Why not just using [file.exists()](https://nodemcu.readthedocs.io/en/master/en/modules/file/#fileexists)?

Maybe I miss something but as far as I can see that construction is going use more CPU in the end, and probably create some object in memory that we will need to trash on the next GC

marcoskirsch, 2017-09-05:
> Why not just using file.exists()?

When I started working on this, in the early days of NodeMCU, there was no exists() method. 

It's a good catch. If you submit a PR to fix it I'll accept right away. Thanks!

Godzil, 2017-09-05:
Hehe :)
That is true that NodeMCU had quite a lot of changes in the default API recently (for the bad and the good)

closed 2018-01-15 by marcoskirsch