marcoskirsch/nodemcu-httpserver #113: Variables that should be local, and a potential "leak":

Godzil, 2017-09-05:
I think I found a few variable that should be local an not global, but maybe I miss something and it is done on purpose:
https://github.com/marcoskirsch/nodemcu-httpserver/blob/b5baf6cd485eec94600f9f782c0fae0988ffc693/httpserver-b64decode.lua#L40
https://github.com/marcoskirsch/nodemcu-httpserver/blob/b5baf6cd485eec94600f9f782c0fae0988ffc693/httpserver-b64decode.lua#L47
https://github.com/marcoskirsch/nodemcu-httpserver/blob/b5baf6cd485eec94600f9f782c0fae0988ffc693/httpserver-request.lua#L82
https://github.com/marcoskirsch/nodemcu-httpserver/blob/b5baf6cd485eec94600f9f782c0fae0988ffc693/httpserver-request.lua#L114

These objects as we return them, I think they are not meant to be global:
https://github.com/marcoskirsch/nodemcu-httpserver/blob/b5baf6cd485eec94600f9f782c0fae0988ffc693/httpserver-basicauth.lua#L5
https://github.com/marcoskirsch/nodemcu-httpserver/blob/b5baf6cd485eec94600f9f782c0fae0988ffc693/httpserver-connection.lua#L8

Here because of the ';', 'i' is a global:
https://github.com/marcoskirsch/nodemcu-httpserver/blob/b5baf6cd485eec94600f9f782c0fae0988ffc693/httpserver-request.lua#L24

One note about httpserver.lua as we voluntarily use two global ( fullPayload and bBodyMissing) because POST data can come in multiple chunk, but does that could happen with chunk sent with different connections? (i.e. the browser connect to the socket, send the first chunk, then disconnect, reconnect again send second chunk etc... ?)

If not, I would make sure in the onDisconnect event to clear them to nil before we call the GC in case of a network issue and the client got disconnected before sending the whole packet and make sure they are not using space in global.

HHHartmann, 2017-09-09:
Hi,

great idea reading the code and finding some issues. I have been wondering about the BufferedConnection and basicAuth variables also and changed them to local on my machine which still seems to work.
Speaking of basicAuth: in httpserver.lua it is assigned to a local variable but this could be freed right after the authenticate call.

As to the collection of chunked post data. I would think that if the connection is lost the request is invalid, so no reconnect and sending further data. But I am not 100% sure about that. But another problem arises from these global variables.
What if we have 2 connections at the same time sending chunked date? it would all get mixed up.

But why not prepare a PR for the obvious things.

Fractal147, 2018-01-05:
Hi, 
I'm using nodemcu-httpserver, and thought I'd start by contributing with some low hanging fruit.
I've confirmed by looking at the globals (_G) table that these do indeed stay around after a connection is closed (at least bufferedConnection, i, and _).
See PR #122!

marcoskirsch, 2018-01-15:
@Godzil this is a nice cleanup, I'm going to close the issue unless you or @HHHartmann tell me not to due to some unfinished detail.

closed 2018-01-15 by marcoskirsch