marcoskirsch/nodemcu-httpserver #119: Allow upload of all files instead of only the http/ files

HHHartmann, 2017-10-17:
I am changing the upload.lua/html to allow up/downloading all files and not only the ones starting with http/.

Are you interested in a PR or have any comments?

marcoskirsch, 2017-10-17:
Is this a wise thing to do, security wise?
Not like stuff under http/ can't do whatever it pleases anyway.

If you see no problem with it, then I'll merge the PR.

HHHartmann, 2017-10-17:
Well I had the same thoughts and reached to the same conclusion.
Maybe I could implement a toggle in the root dir, so admins could ether allow full control or restricted.
But where to store it. maybe in a file like upload.conf or upload.conf.lua or upload.lua?
maybe a key like conf.upload.restrict in httpserver-conf.lua? 
The problem is that httpserver-conf.lua is part of the server and maybe should not be used for application level configuration.

But maybe that would be overdone, since as you said stuff under http can do anything anyways.

The only sensible data on the system is the WiFi password, especially that in STATION mode.

PR #120 (2018-01-15): Upload for all files fix 119
Allow uploading files to http and root dir. fix #119
closed 2018-01-15 by marcoskirsch