marcoskirsch/nodemcu-httpserver #127: nodemcu-uploader doesn't work with latest firmware

marcoskirsch, 2019-11-14:
Makefile uses nodemcu-uploader (https://github.com/kmpm/nodemcu-uploader) but the tool no longer works.
We need to either replace the tool or work on a fix!

seregaxvm, 2019-11-15:
Unfortunately, [`nodemcu-uploader`](https://github.com/kmpm/nodemcu-uploader/) is not being actively maintained right now (<https://github.com/kmpm/nodemcu-uploader/issues/90>).
There's a [`nodemcu-tool`](https://github.com/AndiDittrich/NodeMCU-Tool) which can upload files to nodemcu. However, `nodemcu-httpserver` relies on `nodemcu-uploader` file renaming feature in order to simulate `http` folder in SPIFFS. `nodemcu-tool` does not have this feature. On the bright side, `nodemcu-tool` can execute files (<https://github.com/AndiDittrich/NodeMCU-Tool#5-run-it-directly-and-view-the-output>), so some init code may be run this way.

AndiDittrich, 2019-12-23:
in case you want upload files with nodemcu-tool there is the `keeppath` option available.

## Example ##

```
 $ ./nodemcu-tool upload --keeppath http/*
[NodeMCU-Tool]~ Connected 
[device]      ~ Arch: esp8266 | Version: 3.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0 
[NodeMCU-Tool]~ Uploading "http/apple-touch-icon.png" >> "http/apple-touch-icon.png"... 
[connector]   ~ Transfer-Mode: base64 
[NodeMCU-Tool]~ Uploading "http/args.lua" >> "http/args.lua"... 
[NodeMCU-Tool]~ Uploading "http/cars-bugatti.jpg" >> "http/cars-bugatti.jpg"... 
[NodeMCU-Tool]~ Uploading "http/cars-ferrari.jpg" >> "http/cars-ferrari.jpg"... 
[NodeMCU-Tool]~ Uploading "http/cars.html" >> "http/cars.html"... 
[NodeMCU-Tool]~ Uploading "http/cars-lambo.jpg" >> "http/cars-lambo.jpg"... 
[NodeMCU-Tool]~ Uploading "http/cars.lua" >> "http/cars.lua"... 
[NodeMCU-Tool]~ Uploading "http/cars-mas.jpg" >> "http/cars-mas.jpg"... 
[NodeMCU-Tool]~ Uploading "http/cars-mercedes.jpg" >> "http/cars-mercedes.jpg"... 
[NodeMCU-Tool]~ Uploading "http/cars-porsche.jpg" >> "http/cars-porsche.jpg"... 
[NodeMCU-Tool]~ Uploading "http/counter.html" >> "http/counter.html"... 
[NodeMCU-Tool]~ Uploading "http/file_list.lua" >> "http/file_list.lua"... 
[NodeMCU-Tool]~ Uploading "http/garage_door_control.css" >> "http/garage_door_control.css"... 
[NodeMCU-Tool]~ Uploading "http/garage_door_control.html" >> "http/garage_door_control.html"... 
[NodeMCU-Tool]~ Uploading "http/garage_door.html" >> "http/garage_door.html"... 
[NodeMCU-Tool]~ Uploading "http/garage_door.lua" >> "http/garage_door.lua"... 
[NodeMCU-Tool]~ Uploading "http/hello_world.txt" >> "http/hello_world.txt"... 
[NodeMCU-Tool]~ Uploading "http/index.html" >> "http/index.html"... 
[NodeMCU-Tool]~ Uploading "http/node_info.lua" >> "http/node_info.lua"... 
[NodeMCU-Tool]~ Uploading "http/post.lua" >> "http/post.lua"... 
[NodeMCU-Tool]~ Uploading "http/underconstruction.gif" >> "http/underconstruction.gif"... 
[NodeMCU-Tool]~ Uploading "http/upload.css" >> "http/upload.css"... 
[NodeMCU-Tool]~ Uploading "http/upload.html" >> "http/upload.html"... 
[NodeMCU-Tool]~ Uploading "http/upload.js" >> "http/upload.js"... 
[NodeMCU-Tool]~ Uploading "http/upload.lua" >> "http/upload.lua"... 
[NodeMCU-Tool]~ Uploading "http/zipped.html.gz" >> "http/zipped.html.gz"... 
[NodeMCU-Tool]~ Bulk File Transfer complete! 
[NodeMCU-Tool]~ disconnecting 

 $ ./nodemcu-tool fsinfo
[NodeMCU-Tool]~ Connected 
[device]      ~ Arch: esp8266 | Version: 3.0.0 | ChipID: 0xd1aa | FlashID: 0x1640e0 
[device]      ~ Free Disk Space: 3207 KB | Total: 3442 KB | 36 Files 
[device]      ~ Files stored into Flash (SPIFFS) 
[device]      ~  - http/upload.html (895 Bytes) 
[device]      ~  - http/cars-mas.jpg (12011 Bytes) 
[device]      ~  - http/hello_world.txt (290 Bytes) 
[device]      ~  - http/apple-touch-icon.png (4183 Bytes) 
[device]      ~  - http/garage_door.lua (5336 Bytes) 
[device]      ~  - http/cars-porsche.jpg (12299 Bytes) 
[device]      ~  - http/cars-mercedes.jpg (12283 Bytes) 
[device]      ~  - http/cars-bugatti.jpg (12266 Bytes) 
[device]      ~  - http/file_list.lua (1158 Bytes) 
[device]      ~  - http/upload.css (5511 Bytes) 
[device]      ~  - http/garage_door_control.css (1647 Bytes) 
[device]      ~  - http/counter.html (676 Bytes) 
[device]      ~  - http/upload.js (11121 Bytes) 
[device]      ~  - http/cars.html (905 Bytes) 
[device]      ~  - helloworld.lua (382 Bytes) 
[device]      ~  - http/args.lua (1007 Bytes) 
[device]      ~  - http/cars-ferrari.jpg (7492 Bytes) 
[device]      ~  - http/post.lua (1159 Bytes) 
[device]      ~  - http/garage_door.html (952 Bytes) 
[device]      ~  - http/cars.lua (2144 Bytes) 
[device]      ~  - http/garage_door_control.html (2879 Bytes) 
[device]      ~  - http/node_info.lua (1602 Bytes) 
[device]      ~  - http/zipped.html.gz (363 Bytes) 
[device]      ~  - http/index.html (2733 Bytes) 
[device]      ~  - http/upload.lua (2962 Bytes) 
[device]      ~  - http/cars-lambo.jpg (12992 Bytes) 
[device]      ~  - http/underconstruction.gif (2689 Bytes) 
[NodeMCU-Tool]~ disconnecting 

```